ubilab-escape/second-door #25: MQTT communication

TKoe92, 2019-12-11:
Communicate with the Operator:

- [ ] setup WiFi

- [ ] setup MQTT

- [ ] setup topics and tx/rx messages

JayWiz, 2020-01-17:
MQTT still missing entirely for fusebox

JayWiz, 2020-01-17:
SequenceWorkflow("Solve fusebox", [
                    ActivateLaserWorkflow("Laser_lab_room", "7/laser"),
                    Workflow("Fusebox_lab_room", "7/fusebox/laserDetection"),
                    ParallelWorkflow("Solve fuse box", [
                        Workflow("Fusebox_lab_room", "7/fusebox/rewiring0"),
                        Workflow("Fusebox_lab_room", "7/fusebox/rewiring1"),
                        Workflow("Fusebox_lab_room", "7/fusebox/potentiometer")
                    ]),
                    Workflow("Robot_lab_room", "7/robot")
                ])

TKoe92, 2020-01-23:
use mqtt_base subrepo for communication

closed 2020-01-23 by TKoe92